ARMmbed/wifi-ism43362 #2: Drivers eats 111 bytes even if not used

JanneKiiskila, 2018-02-19:
Seems this driver is having something "extra" that should not be compiled in (or linker doesn't know how to remove).  If you compile mbed-os-example-client for ETHERNET - this driver should not even be linked in - but, in reality it seems to as Mbed OS compiles everything - it doesn't know how to drop it all out.

In case of this driver, it's actually quite lot (rest of them seem to be ~8-16 bytes).

Map-file tells this:

```
 .text._ZNK4mbed8CallbackIFvvEE4callEv
                0x0000000000001dfc       0x28 BUILD/K64F/GCC_ARM/easy-connect/wifi-ism43362/ISM43362/ATParser/BufferedSpi/BufferedSpi.o
                0x0000000000001dfc                mbed::Callback<void ()>::call() const
```

So, the buffered SPI gets linked in every time? 

@SeppoTakalo  - any pointers on this? 

Rest of the drivers seem to leave some callback class. ESP8266 for example;
```

                0x0000000000001dc8       0x10 BUILD/K64F/GCC_ARM/easy-connect/esp8266-driver/ESP8266/ESP8266.o
                0x0000000000001dc8                mbed::Callback<void ()>::~Callback()
                0x0000000000001dc8                mbed::Callback<void ()>::~Callback()
```

JanneKiiskila, 2018-02-19:
I wonder if this driver REALLY needs the ATParser (of it's own), should it use the one under Mbed OS? 
Ref: https://github.com/ARMmbed/mbed-os/issues/4040

Mbed OS Master (unfortunatley, master only - coming to 5.8) also supports buffered Serial, that might remove the need for this whole buffered SPI code?

adustm, 2018-02-19:
Hello @JanneKiiskila 

I don't know how to completely remove the ism43362 driver from the link. Let me know if I need to do some changes for that.

> I wonder if this driver REALLY needs the ATParser (of it's own), should it use the one under Mbed OS?

This wifi driver is using SPI 16bytes exchanges. The ATCmdParser from mbed-os is written for files, using  serial UART 8bytes exchanges.  
The ISM43362 wifi does not have an interrupt when datas are received (no oob processing).
If you compare both files (mbed-os/platform/ATCmdParser.cpp and ism43362/ATParser/ATParser.cpp), I don't know how to get rid of the differences.

cc @LMESTM (fyi)

JanneKiiskila, 2018-11-09:
I think this is resolved by some work by the Mbed OS core networking team, @SeppoTakalo - I think this ticket can be closed?

closed 2018-11-09 by JanneKiiskila