ARMmbed/wifi-ism43362 #3: No SSIDsare  detected on DISCO_L475VG_IOT01A

jononi, 2018-02-26:
I've been using the old [WiFi firmware](https://os.mbed.com/teams/ST/code/DISCO_L475VG_IOT01A_wifi/) without any issue but this driver is not detecting any existing WiFi network and the only SSID I've been able to make it connect to is a hotspot WiFi created on my laptop. Not sure what kind of tests you ran on this board.

LMESTM, 2018-02-26:
@jononi  - what application code are you using for this test ?
When moving to this new driver, we recommend to test with corresponding mbed standard code, like https://os.mbed.com/teams/ST/code/mbed-os-example-wifi/
Have you met issues with the above application or a different one ? 
If you could provide application code reference as well as any application logs, that would help.

jononi, 2018-02-26:
I did use the example you are referring to exactly as it is, and the only time when I can get an SSID listed is when I turn on the mobile hotspot on my PC, although there is several other APs with strong signal around (my home network is not detected even when the board is close by).
Is there a timeout period variable for listening to broadcasting APs that can be increased ?

LMESTM, 2018-02-27:
@jononi 
Yes, there is a timeout right here: 
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362Interface.cpp#L25
But I don't think it would change anything, the WIFI module does not wait for the end of this timeout, as it quickly reports the list of found networks whatever the timeout (I've just tried the change). In my office, whatever the timeout the module will report the 8 or 9 available networks around in less then 5 sec.

Nevertheless, I'd be glad to help, so could you activate traces here:
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ATParser/ATParser.cpp#L39
by defining
```
#define dbg_on 1
#define TRACE_AT_DATA 1
```
then share the logs ?

adustm, 2018-02-27:
Hello @jononi,
Another try would be to comment the scan_demo call and see if the rest of the test is passing.
Let us know.

LMESTM, 2018-02-27:
@jononi  Ok - we found out that the module can report access points in different formats.
I had no such AP in my office, but @adustm found one such AP in her office.
Could you check proposed fix  #4 and let us know if this solves your issue ?

jononi, 2018-02-27:
Thank you ST team for your efforts. :clap: I added the changes you made and now I can see all the broadcasting networks around. The one issue I had is that the detected parameters are not correct, and thus, I still can't connect to my home network. Here a snippet from the scan function output:
```
...
Network: Tunisia secured: None BSSID: hX:hX:hX:hx:hx:hx RSSI: hd Ch: hd
...
Network: Jaafar-XPS15-2 secured: WPA BSSID: hX:hX:hX:hx:hx:hx RSSI: hd Ch: hd
...
```

The first one is for my home network, the name is truncated and its WPA security not detected here.
The second is the mobile hotspot that I create on my PC and its security is correctly detected.

It came to my mind that the space in my SSID name could be a problem so I created a guest wifi network with no spaces in its name and same security setting. (and by the way, you still have that security setting hard coded to WPA2 in [here](https://github.com/ARMmbed/wifi-ism43362/blob/e4ecc27e87d96072f7df62a25ef007986dc95c4e/ISM43362/ISM43362.cpp#L190)

this time the new SSID is detected: 
```
...
Network: Dothraki secured: WPA BSSID: hX:hX:hX:hx:hx:hx RSSI: hd Ch: hd
...
```
And the wifi module connected to it with no problems.

So you can add a new issue for not connecting to SSIDs with space character in their names. :smile:

adustm, 2018-02-27:
Hello @jononi 
Pr#5 fixes the security level programmation

adustm, 2018-02-27:
Hello @jononi 
I have used 'Something Tologon' as SSID, and I can see the space character is well transmitted to the wifi module (you can see it when using the trace tips from @LMESTM 
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ATParser/ATParser.cpp#L39
by defining
```
#define dbg_on 1
#define TRACE_AT_DATA 1
```
)

There will be a moment where you will read in the Hyper Terminal 
AT> C1=Something Tologon

I will have to create a wifi AP with a space in it to check that the connection is passing. By the way I cannot see a space in the wifi AP that you have shown:

> Network: Tunisia secured: None BSSID: hX:hX:hX:hx:hx:hx RSSI: hd Ch: hd
> ...
> Network: Jaafar-XPS15-2 secured: WPA BSSID: hX:hX:hX:hx:hx:hx RSSI: hd Ch: hd
> 

Kind regards

jononi, 2018-02-27:
I enabled the tracing log and created another network named: "guest network"
Here's what I got:

```
...
AT? ====#%*s
%n====
23 30 30 31 2C 22 67 75 65 73 74 20 AT= ====#001,"guest ====
Pending data
AT? ====#%*s
%n====
6E 65 74 77 6F 72 6B 22 2C 35 34 3A 41 30 3A 35 30 3A 44 31 3A 46 34 3A 41 39 2C 2D 37 34 2C 32 31 36 2E 37 2C 49 6E 66 72 61 73 74 72 75 63 74 75 72 65 2C 4F 70 65 6E 2C 32 2E 34 47 48 7A 2C 35  D  A New line AT<<< network",54:A0:50:D1:F4:A9,-74,216.7,Infrastructure,Open,2.4GHz,5
...

```
It looks like the space in the ssid name is creating the confusion that there are two separate AP, one with ssid="guest" and the other ssid="network". 

Looking at the connection attempt, here the output:

```
AT> C1=guest network

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C2=

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C3=0

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C0

NO DATA, read again
AT? ====OK
%n====
41 6C 72 65 61 64 79 20 63 6F 6E 6E 65 63 74 65 64 21  D  A New line AT<<< Already connected!
45 52 52 4F 52 3A 20 55 6E 6B 6E 6F 77 6E 20 45 72 72 6F 72  D  A New line AT<<< ERROR: Unknown Error
55 73 61 67 65 3A 20 43 30 20  D  A New line AT<<< Usage: C0 
3E 20 AT(Timeout)
```

Like you mentioned, the ssid is entered correctly with C1 command, same thing for the security level (C3=0, open network), but there's no way to get successful connection. 
I tried again with WPA2 security this time. this is the full log output:

```
> Connecting to guest network...
AT> I?

NO DATA, read again
AT? ====%*s
%n====
49 53 4D 34 33 33 36 32 2D 4D 33 47 2D 4C 34 34 2D 53 50 49 2C 43 33 2E 35 2E 32 2E 33 2E 42 45 54 41 39 2C 76 33 2E 35 2E 32 2C 76 31 2E 34 2E 30 2E 72 63 31 2C 76 38 2E 32 2E 31 2C 31 32 30 30 30 30 30 30 30 2C 49 6E 76 65 6E 74 65 6B 20 AT= ====ISM43362-M3G-L44-SPI,C3.5.2.3.BETA9,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek ====
Pending data
AT? ====OK
%n====
65 53 2D 57 69 46 69  D  A New line AT<<< eS-WiFi
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C4=1

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C1=guest network

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C2=stm32l475

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C3=3

NO DATA, read again
AT? ====OK
%n====
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C0

NO DATA, read again
AT? ====OK
%n====
5B 4A 4F 49 4E 20 20 20 5D 20 67 75 65 73 74 20 6E 65 74 77 6F 72 6B 2C 31 39 32 2E 31 36 38 2E 32 39 2E 31 36 36 2C 30 2C 30  D  A New line AT<<< [JOIN   ] guest network,192.168.29.166,0,0
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
AT> C?

NO DATA, read again
AT? ====%*s
%n====
67 75 65 73 74 20 AT= ====guest ====
Pending data
AT? ====OK
%n====
6E 65 74 77 6F 72 6B 2C 73 74 6D 33 32 6C 34 37 35 2C 33 2C 31 2C 30 2C 31 39 32 2E 31 36 38 2E 32 39 2E 31 36 36 2C 32 35 35 2E 32 35 35 2E 32 35 35 2E 30 2C 31 39 32 2E 31 36 38 2E 32 39 2E 31 2C 32 35 35 2E 32 35 35 2E 32 35 35 2E 32 35 35 2C 32 35 35 2E 32 35 35 2E 32 35 35 2E 32 35 35 2C 33 2C 30 2C 30 2C 55 53 2C 31  D  A New line AT<<< network,stm32l475,3,1,0,192.168.29.166,255.255.255.0,192.168.29.1,255.255.255.255,255.255.255.255,3,0,0,US,1
4F 4B  D  A AT= ====OK
====
Pending data
AT? ====> 
%n====
3E 20 AT= ====> ====
> Cannot connect to AP.
```

Let me know if you can reproduce this issue. For now I'm ok working around this limitation.

LMESTM, 2018-02-28:
@jononi Thanks
I created issue #6 to track space related issue. There seems to be a problem with the AT parser that stops at the space character instead of the end of line. So let's track the space issue in separate thread.

adustm, 2018-04-03:
Hello @jononi @LMESTM 
Do you agree to close this issue ?

jononi, 2018-04-04:
SSID name with space is no longer an issue now. I confirm it can be listed and joined.

closed 2018-04-04 by jononi