ARMmbed/wifi-ism43362 #6: Scan failes if Access Point names contains a space

LMESTM, 2018-02-28:
As reported in initial issue #3 

from @jononi
I enabled the trace log and created another network named: "guest network"
Here's what I got:

...
AT? ====#%*s
%n====
23 30 30 31 2C 22 67 75 65 73 74 20 AT= ====#001,"guest ====
Pending data
AT? ====#%*s
%n====
6E 65 74 77 6F 72 6B 22 2C 35 34 3A 41 30 3A 35 30 3A 44 31 3A 46 34 3A 41 39 2C 2D 37 34 2C 32 31 36 2E 37 2C 49 6E 66 72 61 73 74 72 75 63 74 75 72 65 2C 4F 70 65 6E 2C 32 2E 34 47 48 7A 2C 35  D  A New line AT<<< network",54:A0:50:D1:F4:A9,-74,216.7,Infrastructure,Open,2.4GHz,5
...

It looks like the space in the ssid name is creating the confusion that there are two separate AP, one with ssid="guest" and the other ssid="network".

LMESTM, 2018-02-28:
There seems to be a problem with the AT parser that stops at the space character instead of the end of line - to be investigated

adustm, 2018-04-03:
Hello @jononi
PR#8 fixes this issue.

Let us know if it's ok and we'll close this one.

jononi, 2018-04-04:
It's fixed now. Thanks

closed 2018-04-05 by adustm