ARMmbed/wifi-ism43362 #15: Driver does not define SPI pins, but depend on application to define those

SeppoTakalo, 2018-05-31:
This driver does not correctly define its build time dependencies.

https://github.com/ARMmbed/wifi-ism43362 driver requires SPI pin configurations, it should therefore have `mbed_lib.json` that defines these variables, and use macros generated from those.
Defaults should be `NC` so that it builds on all platforms, them you can provide sensible defaults for some platforms you know the pin names already.

This mechanism you are now using is a bit upside down approach. Your driver depend on application to define build time macros. Effectively breaking build for every platform that adds this driver.

jeromecoutant, 2018-05-31:
I tried to implement the correct mechanism.
Could you check https://github.com/jeromecoutant/wifi-ism43362/tree/PR_JSON
Thx

SeppoTakalo, 2018-06-01:
I'll close this,

I misunderstood the documentation.
The driver is actually NOT using those configuration values, but instead expecting application to use these. Therefore the `README.md` is misleading.

closed 2018-06-01 by SeppoTakalo