ARMmbed/wifi-ism43362 #24: ATParser.cpp: incompatible redefinition of macro MIN

teetak01, 2018-06-21:
### Description
Mbed OS 5.9.1
Component: ARM Compiler 5.06 update 3 (build 300)

```
ATParser.cpp@37,0:  #47-D: incompatible redefinition of macro "MIN" (declared at line 145 of "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_common.h")
```
[ ] Question  
[ ] Enhancement  
[X] Bug  

This driver is still using its own ATParser.cpp. Would it make sense to use the Mbed OS implementation instead and remove this component?

For. ESP8266 removed the dependency here: https://github.com/ARMmbed/esp8266-driver/pull/46

teetak01, 2018-06-21:
Originally reported here: https://github.com/ARMmbed/mbed-os/issues/7256

0xc0170, 2019-02-19:
@jeromecoutant Is this still an issue? Shall be closed or fixed?

jeromecoutant, 2019-02-19:
About MIN macro redefinition, to be honest, I don't know

About using AT parser from mbed-os, this is still in my long todo list :-)

closed 2021-12-02 by teetak01