ARMmbed/wifi-ism43362 #26: Running WiFi example on DISCO_F413ZH board

danny4478, 2018-08-09:
I am trying to run the WiFi example https://github.com/ARMmbed/mbed-os-example-wifi on DISCO_F413ZH board, but the scan does not find any access points.
wifi->scan() returns 0 and I get “scan() failed with return value: 0”.
 
When ignoring the scan failure and attempt to connect I get this error:, 
NSAPI_ERROR_DEVICE_ERROR [failure interfacing with the network processor (-3012)]

Am I using a supported board? See pictures below.
Any other idea why this scan fails?

[image]
[image]

jeromecoutant, 2018-08-09:
Hi

Yes, DISCO_F413ZH is a supported board,
and yes we have some issues with the scan procedure when wifi environment is very populated...

Is it working if you skip the full scan_demo part ?

danny4478, 2018-08-09:
No.
When ignoring the scan failure and attempt to connect I get this error:
NSAPI_ERROR_DEVICE_ERROR [failure interfacing with the network processor (-3012)]

LMESTM, 2018-08-14:
@danny4478  what jerome proposed as a test is to **skip the entire scan** part and directly go to the connect - have you tried that ?  There is a known issue in the firmware when too many APs around.

danny4478, 2018-08-15:
Hi
After removing the entire scan part, and directly go to the connect as suggested, we get a successful connection.
@jeromecoutant @LMESTM Thanks for you support on this.

We are now trying to run this example on a Cypress PSoC 6 BT-WiFi board. Assuming we manage to add this new target to mbed-os, is there any reason this wifi example would not work on this board?

jeromecoutant, 2018-09-11:
Hi
Maybe we can close this issue ?
Thx

danny4478, 2018-09-12:
I have closed it.
Thanks

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

closed 2018-09-12 by danny4478