ARMmbed/wifi-ism43362 #41: IPv6 address returned instead of IPv4 in an mbed-os DNS greentea test suite

michalpasztamobica, 2019-03-26:
When running the tests-netsocket-dns test suite I came across a repeatable failure in ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC:
```
[1553519786.14][CONN][RXD] >>> Running case #5: 'ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC'...
[1553519786.15][CONN][INF] found KV pair in stream: {{__testcase_start;ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC}}, queued...
[1553519786.17][CONN][RXD] DNS: query "google.com" => "172.217.16.46"
[1553519786.43][CONN][WRN] UnicodeDecodeError encountered!
[1553519786.43][CONN][RXD] DNS: query "youtube.com" => ":"
mbedgt: :43::FAIL: Expected 0 Was -3009
[1553519786.43][CONN][RXD] :43::FAIL: Expected 0 Was -3009
mbedgt: :44::FAIL: Expression Evaluated To FALSE
[1553519786.43][CONN][RXD] :44::FAIL: Expression Evaluated To FALSE
[1553519786.65][CONN][WRN] UnicodeDecodeError encountered!
[1553519786.65][CONN][RXD] DNS: query "facebook.com" => ":"
mbedgt: :43::FAIL: Expected 0 Was -3009
[1553519786.67][CONN][RXD] :43::FAIL: Expected 0 Was -3009
mbedgt: :44::FAIL: Expression Evaluated To FALSE
[1553519786.67][CONN][RXD] :44::FAIL: Expression Evaluated To FALSE
[1553519786.86][CONN][RXD] DNS: query "wikipedia.org" => "91.198.174.192"
```
Finding the ":" indicates that an IPv6 address has been returned from DNS, rather than IPv4. I have checked that mbed-os is passing an IPv4 version of the address when calling sendto. 
I can only see this failure for DISCO_L475VG_IOT01A and DISCO_F413ZH which are the two boards using the ISM43362 driver. All other boards are passing fine.

Step to reproduce:
1) Compile mbed-os tests (compiler doesn't matter, IAR is just an example)
```
mbed test --compile -t IAR -m DISCO_L475VG_IOT01A -n tests-netsocket-dns --app-config=mbed_app.wlan0.json -v
```
2) Run the compiler test suite:
```
mbed test --run -t IAR -m DISCO_L475VG_IOT01A -n tests-netsocket-dns -v
```

mbed-os was compiled to use Wi-Fi, LWIP and IPv4:
```
 "target_overrides": {
        "*": {
            "target.network-default-interface-type": "WIFI",
            "nsapi.default-stack": "LWIP",
            "lwip.ipv4-enabled": true,
            "lwip.ipv6-enabled": false,
        },
        "DISCO_L475VG_IOT01A": {
            "ism43362.provide-default": true
        }

```

jeromecoutant, 2019-03-26:
Hi

I confirm that with 5.12.0 release, I got the same issue with this test .

(test was OK with 5.11.5)

jeromecoutant, 2019-03-26:
But I don't think it is related to IPV6

Here is 3 different output I can see in ST CI for this tests:

````
[1553599853.49][CONN][RXD] DNS: query "youtube.com" => "D."
mbedgt: :43::FAIL: Expected 0 Was -3009
````
````
[1553600697.75][CONN][RXD] DNS: query "youtube.com" => "	"
mbedgt: :43::FAIL: Expected 0 Was -3009
````
````
[1553599009.23][CONN][RXD] DNS: query "youtube.com" => ":"
mbedgt: :43::FAIL: Expected 0 Was -3009
````

michalpasztamobica, 2019-03-26:
@jeromecoutant , thanks for your quick response!
I can see the following in our nightly CI builds:
```
[1553574903.19][CONN][RXD] >>> Running case #5: 'ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC'...
[1553574903.19][CONN][INF] found KV pair in stream: {{__testcase_start;ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC}}, queued...
[1553574903.29][CONN][RXD] DNS: query "google.com" => "172.217.20.46"
[1553574903.59][CONN][RXD] DNS: query "youtube.com" => "2001:1608:10:25::1c04:b12f"
[1553574903.59][CONN][RXD] :43::FAIL: Expected 0 Was -3009
[1553574903.69][CONN][RXD] :44::FAIL: Expression Evaluated To FALSE
```
Locally I always get the ":" parsing error, so I though it could be something with ipv6, 
However looking at your outputs - this could be just me always getting the same random memory corruption. I will adjust the title, to make it more generic, until the root cause is known...

jeromecoutant, 2019-03-26:
Test is failing since https://github.com/ARMmbed/mbed-os/pull/9387

@tymoteuszblochmobica could you help finding the issue in wifi-ism43362 ?

@SeppoTakalo 

Thx

michalpasztamobica, 2019-03-26:
I guess the [ESP8266 stack](https://github.com/ARMmbed/mbed-os/tree/master/components/wifi/esp8266-driver) should work in a similar manner to ISM43362?
Merging multihoming did not affect ESP8266 stack, so perhaps a comparison of the two would reveal something?
I can see [ESP8266 uses the default gethostbyname()](https://github.com/ARMmbed/mbed-os/blob/ca62922c9fb261d98ca2fbe7533f888a81e474c6/components/wifi/esp8266-driver/ESP8266Interface.h#L174) implementation, for example. It is also aware of UDP and TCP when performing send, which might perhaps affect DNS resolution?

michalpasztamobica, 2019-06-06:
Hello :) Is there any progress on this?

TuomoHautamaki, 2019-06-18:
Hi @jeromecoutant could you help to resolve this with Michal.

michalpasztamobica, 2019-08-01:
I took another look at this issue. Let me mention my findings and thoughts, in case they turn out to be helpful.

1) I reverted to mbed-os rev. [7201cab3c48f3b4f3fb91c69c22e519c631f562e](https://github.com/ARMmbed/mbed-os/commit/7201cab3c48f3b4f3fb91c69c22e519c631f562e) which is before https://github.com/ARMmbed/mbed-os/pull/9387 and I got the same failure as described above.
```
[1564661548.25][CONN][RXD] DNS: query "youtube.com" => "
                                                                "
mbedgt: :43::FAIL: Expected 0 Was -3009
[1564661548.28][CONN][RXD] :43::FAIL: Expected 0 Was -3009
mbedgt: :44::FAIL: Expression Evaluated To FALSE
[1564661548.34][CONN][RXD] :44::FAIL: Expression Evaluated To FALSE
```

2) the test which is failing ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC simultaneously  opens two sockets: one for the asynchronous DNS query and another one for the synchronous one. In this manner it differs from other tests. After enabling debug traces for the drives I can see there is some issue with using the sockets:
```
[1564653011.18][CONN][INF] found KV pair in stream: {{__testcase_name;ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC}}, queued...
[1564653011.25][CONN][RXD] >>> Running case #1: 'ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC'...
[1564653011.31][CONN][INF] found KV pair in stream: {{__testcase_start;ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC}}, queued...
[1564653011.36][CONN][RXD]      ISM43362: receivedIPAddress: 192.168.0.106
[1564653011.41][CONN][RXD]      ISM43362: receivedIPAddress: 192.168.0.106
[1564653011.45][CONN][RXD] ISM43362Interface: socket_open id=0 proto=1
[1564653011.50][CONN][RXD] ISM43362Interface: socket_open id=1 proto=1
[1564653011.53][CONN][RXD] ISM43362Interface: socket_attach id 0
[1564653011.58][CONN][RXD] ISM43362Interface: socket_attach id 1
[1564653011.68][CONN][RXD]      ISM43362: open ok with id 0 type 1 addr 8.8.8.8 port 53
[1564653011.74][CONN][RXD] ISM43362Interface socket_send_nolock id 0 size 31
[1564653011.77][CONN][RXD]      ISM43362 send: id 0 amount 31
[1564653011.81][CONN][RXD]      ISM43362:
[1564653011.86][CONN][RXD]      ISM43362: Buffer content ==========
[1564653011.88][CONN][RXD]      ISM43362: open: P6 issue
[1564653011.92][CONN][RXD] ISM43362Interface socket_check_read: i 0
[1564653011.97][CONN][RXD]      ISM43362 check_recv_status: id 0
[1564653012.02][CONN][RXD]      ISM43362 check_recv_status: id 0 read_amount=47
[1564653012.08][CONN][RXD] ISM43362Interface socket_check_read read_amount 47
[1564653012.12][CONN][RXD]      ISM43362:
[1564653012.16][CONN][RXD]      ISM43362: Buffer content ==========
[1564653012.19][CONN][RXD]      ISM43362: open: P3 issue
[1564653012.26][CONN][RXD] ISM43362Interface socket_recv: req=512 read_data_size=47 connected 1
[1564653012.33][CONN][RXD] ISM43362Interface socket_recv: recv=47
[1564653012.34][CONN][RXD]      ISM43362:
[1564653012.37][CONN][RXD]      ISM43362: Buffer content ==========
[1564653012.40][CONN][RXD]      ISM43362: open: P6 issue
[1564653012.45][CONN][RXD]      ISM43362:
[1564653012.49][CONN][RXD]      ISM43362: Buffer content ==========
[1564653012.52][CONN][RXD]      ISM43362: open: P6 issue
[1564653012.59][CONN][RXD] ISM43362Interface socket_recv: req=512 read_data_size=0 connected 1
[1564653012.62][CONN][RXD]      ISM43362 check_recv_status: id 0
[1564653012.69][CONN][RXD]      ISM43362 check_recv_status: id 0 read_amount=0
[1564653012.75][CONN][RXD] ISM43362Interface socket_recv: returns WOULD BLOCK
[1564653012.78][CONN][RXD]      ISM43362:
[1564653012.81][CONN][RXD]      ISM43362: Buffer content ==========
[1564653012.84][CONN][RXD]      ISM43362: open: P3 issue
[1564653012.89][CONN][RXD] ISM43362Interface socket_check_read: i 0
[1564653012.92][CONN][RXD]      ISM43362 check_recv_status: id 0
[1564653012.97][CONN][RXD]      ISM43362 check_recv_status: id 0 read_amount=0
[1564653013.02][CONN][RXD] ISM43362Interface: socket_attach id 1
[1564653013.05][CONN][RXD] ISM43362Interface: socket_attach id 0
[1564653013.10][CONN][RXD] ISM43362Interface: socket_close, id=1
[1564653013.13][CONN][RXD]      ISM43362: CLOSE socket id=1
[1564653013.19][CONN][RXD] DNS: queISM43ry "g362Iooglenterf.com"ace:  => socke""
[1564653013.22][CONN][RXD] :t_clo43::Fse, AIL: id=0
[1564653013.26][CONN][RXD] Expec        ISM4ted 3362:0 Was CLOS -300E so9
[1564653013.27][CONN][RXD] :44cket ::FAIid=0
[1564653013.30][CONN][RXD] L: Express!!!!!ion E
[1564653013.32][CONN][RXD] valuated To FALSE
mbedgt: :45::FAIL: Expression Evaluated To FALSE
[1564653013.36][CONN][RXD] :45::FAIL: Expression Evaluated To FALSE
```

@jeromecoutant , could it be that it is not possible to open two sockets at a time with the ism43362 wifi driver? I don't see any explicit reason for this in the code.

jeromecoutant, 2019-08-21:
Hi
Tests with #61 are on going

jeromecoutant, 2019-08-21:
Closing as test should be OK now

PR #44 (2019-07-03): Use the default gethostbyname implementation
Code is removes as they were no more called any way...

NB: this doesn't fix #41 

@michalpasztamobica
@SeppoTakalo 
@LMESTM
closed 2019-08-21 by jeromecoutant