ARMmbed/wifi-ism43362 #42: Sending in quick succession fails

janjongboom, 2019-03-26:
Here I have a large message that is sent, which is then automatically split up in multiple packets. However the second one will fail immediately:

```
ISM43362Interface::send length 4168
ISM43362Interface::send length 2708
ISM43362Interface: socket_send ERROR
```

Adding a 50 ms. sleep in send() fixes it, but that shouldn't be a solution. Not sure how to fix this @jeromecoutant

janjongboom, 2019-03-26:
I did it like this now:

```diff
@@ -417,8 +421,14 @@ int ISM43362Interface::socket_send_nolock(void *handle, const void *data, unsign
 {
     struct ISM43362_socket *socket = (struct ISM43362_socket *)handle;

+    // https://github.com/ARMmbed/wifi-ism43362/issues/42
+    // if the size is too big then the packet will be split up,
+    // and the radio module will throw an error when sending too fast
+    bool pause_after_sending = false;
+
     if (size > ES_WIFI_MAX_TX_PACKET_SIZE) {
         size = ES_WIFI_MAX_TX_PACKET_SIZE;
+        pause_after_sending = true;
     }

     if (!_ism.send(socket->id, data, size)) {
@@ -426,6 +436,10 @@ int ISM43362Interface::socket_send_nolock(void *handle, const void *data, unsign
         return NSAPI_ERROR_DEVICE_ERROR; // or WOULD_BLOCK ?
     }

+    if (pause_after_sending) {
+        wait_ms(50);
+    }
+
     return size;
 }
```

LMESTM, 2019-03-26:
@janjongboom 
thanks for the proposed fix. 
One point first - have you checked which Inventek firmware version is on your board? 
C3.5.2.5.STM version would be great and there is a small (Windows) utility and readme here: 
https://www.st.com/resource/en/utilities/inventek_fw_updater.zip
The version can be checked on console when activating 
#define ism_debug 1
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ISM43362.cpp#L22

janjongboom, 2019-03-26:
@LMESTM Yeah, testing it now. This is my log:

```
[INFO][TLSW]: No verification issue for this certificate

ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
socket_send size 4168
ISM43362 send: id 0 amount 1460
socket_send ret 1460
socket_send size 2708
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
ISM43362 send: id 0 amount 1460
socket_send ret 1460
socket_send size 1248
ISM43362: 
ISM43362: Buffer content ==========
ISM43362Interface: socket_send ERROR
socket_send ret -3012
[ERR ][TLSW]: Socket send error -3012
```

janjongboom, 2019-03-26:
Full log with updated module:

```
Opening network interface...
ISM43362: Reset Module
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3525
ISM43362: connection_status 3
ISM43362Interface: update_conn_state_cb 2 -> 3
ISM43362: connection_status 1
ISM43362Interface: update_conn_state_cb 3 -> 1
ISM43362Interface: connect_status 0
ISM43362: receivedIPAddress: 192.168.1.13
ISM43362: receivedIPAddress: 192.168.1.13
Connected to the network successfully. IP address: 192.168.1.13

ISM43362 dns_lookup: 216.239.35.8 ok
ISM43362Interface: socket_open, id=0
ISM43362: open ok with id 0 type 1 addr 216.239.35.8 port 123
ISM43362 send: id 0 amount 48
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
ISM43362Interface socket_recv: returns WOULD BLOCK
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=48
ISM43362Interface socket_recv: recv=48
ISM43362Interface: socket_close, id=0
ISM43362: CLOSE socket id=0
Time is now Tue Mar 26 17:25:52 2019
Thread: 0x20003EA4, Stack size: 2184 / 4096
Thread: 0x200025A0, Stack size: 88 / 512
Thread: 0x20001104, Stack size: 136 / 2048
Thread: 0x2000255C, Stack size: 104 / 768
Heap size: 13524 / 81296 bytes (max: 15000 bytes)
Connecting to host jan123.azure-devices.net:8883 ...
ISM43362Interface: socket_open, id=0
[DBG ][TLSW]: mbedtls_ssl_conf_ca_chain()
ISM43362 dns_lookup: 51.144.118.31 ok
ISM43362: open ok with id 0 type 0 addr 51.144.118.31 port 8883
[INFO][TLSW]: Starting TLS handshake with jan123.azure-devices.net
[DBG ][TLSW]: mbedtls_ssl_setup()
socket_send size 153
ISM43362 send: id 0 amount 153
sockISM43362 check_recv_status: id 0
ISM43362 checket_send ret 153
_recv_status: id 0 read_amount=0
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=1200
ISM43362Interface socket_recv: recv=5
ISM43362Interface socket_recv: recv=1195
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=1200
ISM43362Interface socket_recv: recv=1200
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=1200
ISM43362Interface socket_recv: recv=1200
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=615
ISM43362ISM43362 check_recv_status: id 0
ISM43362 checkInterface socket_recv: recv=615
_recv_status: id 0 read_amount=0
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
ISM43362 check_recv_status: id 0
recv_status: id 0 read_amount=0
[2K[DBG ][TLSW]: 
Verifying certificate at depth 2:

[DBG ][TLSW]:   cert. version     : 3
  serial number     : 02:00:00:B9
  issuer name       : C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
  subject name      : C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
  issued  on        : 2000-05-12 18:46:00
  expires on        : 2025-05-12 23:59:00
  signed using      : RSA with SHA1
  RSA key size      : 2048 bits
  basic constraints : CA=true, max_pathlen=3
  key usage         : Key Cert SISM43362 check_recv_status: id 0
ISM43362 checkign, CRL Sign

[INFO][TLSW]: No verificati_recv_status: id 0 read_amount=0
on issue for this certificate

[DBG ][TLSW]: 
Verifying certificate at depth 1:

[DBG ][TLSW]:   cert. version     : 3
  serial number     : 0F:2C:10:C9:5B:06:C0:93:7F:B8:D4:49:F8:3E:85:69
  issuer name       : C=IE, O=Baltimore, OU=CyberTrust, CN=Baltimore CyberTrust Root
  subject name      : C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, OU=Microsoft IT, CN=Microsoft IT TLS CA 2
  issued  on        : 2016-05-20 12:51:57
  expires on        : 2024-05-20 12:51:57
  signed using      : RSA with ISM43362 check_recv_status: id 0
ISM43362 checkSHA-256
  RSA key size      : 4096 bits
  basic constrai_recv_status: id 0 read_amount=0
nts : CA=true, max_pathlen=0
  key usage         : Digital Signature, Key Cert Sign, CRL Sign
  ext key usage     : TLS Web Server Authentication, TLS Web Client Authentication, OCSP Signing

[INFO][TLSW]: No verification issue for this certificate

[DBG ][TLSW]: 
Verifying certificate at depth 0:

[DBG ][TLSW]:   cert. version     : 3
  serial number     : 20:00:05:A3:2E:55:47:67:A5:DA:A8:3A:28:00:00:00:05:A3:2E
  issuer name       : C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, OUISM43362 check_recv_status: id 0
ISM43362 check=Microsoft IT, CN=Microsoft IT TLS CA 2
  subject name  _recv_status: id 0 read_amount=0
    : CN=*.azure-devices.net
  issued  on        : 2019-03-14 14:46:08
  expires on        : 2021-03-14 14:46:08
  signed using      : RSA with SHA-256
  RSA key size      : 2048 bits
  subject alt name  : *.azure-devices.net, *.amqpws.azure-devices.net, *.su.management-azure-devices.net
  key usage         : Digital Signature, Key Encipherment, Data Encipherment
  ext key usage     : TLS Web Client Authentication, TLS Web Server Authentication

[INFO][TLSW]: No verification issue for this certificate

ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
socket_send size 4168
ISM43362 send: id 0 amount 1460
socket_send ret 1460
socket_send size 2708
ISM43362 check_recv_status: id 0
ISM43362 check_recv_status: id 0 read_amount=0
ISM43362 send: id 0 amount 1460
socket_send ret 1460
socket_send size 1248
ISM43362: 
ISM43362: Buffer content ==========
ISM43362Interface: socket_send ERROR
```

janjongboom, 2019-03-26:
Actually my proposed fix now stopped working too; wondering what happened.

janjongboom, 2019-03-26:
Here's an example program: https://github.com/coisme/Mbed-to-Azure-IoT-Hub/pull/2

LMESTM, 2019-03-27:
So do you see exactly the same trace when adding back your workaround ? 
Looks like we're not receiving "OK" back from the module. 
Would you run one more test with moving debug to true in BufferedSpi.cpp ? 
   #define local_debug false
This should confirm the timeout happens. I'd like to verify how much the timeout is set. 
It's supposed to be very high here (1 minute) - do you see  a long pause before the Error happens ? 

Would need more time to reproduce the issue locally, as I guess it requires to setup the Azure account with credentials and so on ...

janjongboom, 2019-03-27:
@LMESTM Do you have an email address and I can send you a file with my certificate?

janjongboom, 2019-03-28:
Could this be a corruption issue?

```

struct ISM43362_socket {
    int id;
    nsapi_protocol_t proto;
    volatile bool connected;
    SocketAddress addr;
    char read_data[1400];
    volatile uint32_t read_data_size;
};
```

But max data read is 1460 bytes. Potentially writing outside the buffer?

jeromecoutant, 2019-04-03:
Should be fixed with #47

closed 2019-04-03 by jeromecoutant