ARMmbed/wifi-ism43362 #50: mbed-os Greentea tests-network-wifi testsuite fails with AUTH_FAILURE

michalpasztamobica, 2019-04-04:
mbed-os greentea tests-network-wifi is failing on multiple test cases returning AUTH_FAILURE in [this line](https://github.com/ARMmbed/wifi-ism43362/blob/cfbb9766888732b1090b6b69d3772381d87e9acd/ISM43362/ISM43362.cpp#L230):

```
[1554349113.14][CONN][RXD] 
[1554349113.14][CONN][RXD] >>> Running case #8: 'WIFI-GET-RSSI'...
[1554349113.24][CONN][INF] found KV pair in stream: {{__testcase_start;WIFI-GET-RSSI}}, queued...
[1554349141.88][CONN][RXD] :36::FAIL: Expected 0 Was -3011
[1554349142.15][CONN][RXD] :40::FAIL: Expected 0 Was -3004
[1554349142.25][CONN][INF] found KV pair in stream: {{__testcase_finish;WIFI-GET-RSSI;0;2}}, queued...
[1554349142.35][CONN][RXD] >>> 'WIFI-GET-RSSI': 0 passed, 2 failed with reason 'Test Cases Failed'
[1554349142.35][CONN][RXD] 
[1554349142.35][CONN][RXD] >>> Running case #9: 'WIFI-CONNECT-PARAMS-VALID-UNSECURE'...
[1554349142.46][CONN][INF] found KV pair in stream: {{__testcase_start;WIFI-CONNECT-PARAMS-VALID-UNSECURE}}, queued...
[1554349170.77][CONN][RXD] :31::FAIL: Expected 0 Was -3011
[1554349170.77][CONN][RXD] :32::FAIL: Expected 0 Was -3004
[1554349192.66][CONN][RXD] :33::FAIL: Expected 0 Was -3011
[1554349192.75][CONN][RXD] :34::FAIL: Expected 0 Was -3004
```
The issue is 99% reproducible, but I got those tests passing if I commented out the two WIFI-SCAN tests. At first glance I could not find the issue in the code, perhaps a memory corruption somewhere?

This would be in line with the fact that in other test suites, where DISCO simply connects to Wi-Fi AP without scanning, the tests are passing fine.

Another important finding is that these tests only started failing in mbed-os nightly runs two weeks ago. But when I rolled back to mbed-os and ism43326 from before that day I still couldn't pass the wifi suite...

The reason why I decided to file an issue in this project is that we also checked DISCO_F413ZH and it failed in the same way. All other boards connect fine to the Wi-Fi both on our desks and in the RAAS system. Therefore, I would be grateful if you checked this issue.

@TuomoHautamaki , @SeppoTakalo , FYI

jeromecoutant, 2019-04-04:
Hi
Analysis is on going

closed 2019-04-12 by jeromecoutant

michalpasztamobica, 2019-04-15:
@jeromecoutant , I am sorry, but I can still see the same AUTH_FAILURE errors in our Greentea test suite, with the latest revision of wifi-ism43362 driver. 
Would you please double check that the revert which you applied fixes the issue?

michalpasztamobica, 2019-04-23:
@jeromecoutant , please, double check. The issue is not fixed. I can see the failures repeated every day in our CI.

michalpasztamobica, 2019-04-25:
@jeromecoutant , I am sorry, you are right - F413ZH behaves correctly now, so the issues with L475 that we're seeing cannot be related to the wifi driver. We will investigate further on our side. Thanks for checking!

PR #55 (2019-04-12): Revert "Dimension SPI TX Buffer to cope with Max Data"
This reverts commit a66d89d173f3eae08662b8ea89f8af59a641bec6 from #47 

See https://github.com/ARMmbed/wifi-ism43362/pull/47#pullrequestreview-222737246

Issue was seen with DISCO_F413ZH and tests-network-wifi

Fix #50