ARMmbed/wifi-ism43362 #53: WiFi module ISM43362-M3G-L44 scanning was failed , if surrounded access points are more than 20.

thota4u, 2019-04-12:
Hi Team,

We are working on DISCO-L475VG-IOT01A board with WiFi module Inventek ISM43362-M3G-L44 (802.11 b/g/n compliant) for ARM mbed pelion cloud.

issue : WiFi module ISM43362-M3G-L44 scanning was failed , if surrounded access points are more than 20.

could you please let me know how to fix this issue?

Thanks & Regards
Praveen.Thota

jeromecoutant, 2019-04-12:
Hi

One point first - have you checked which Inventek firmware version is on your board?
C3.5.2.5.STM version would be great and there is a small (Windows) utility and readme here:
https://www.st.com/resource/en/utilities/inventek_fw_updater.zip

The version can be checked on console when activating
#define ism_debug 1
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ISM43362.cpp#L22

thota4u, 2019-04-12:
Hi Jerome Coutant,

Thanks for your reply.

yes , we already updated our firmware version to C3.5.2.5.STM version 
still we are facing the issue on mbed os 5.10.4 : scan fails if access points are more than 20.

We also tried the same board with Amazon Free RTOS with latest firmware version : C3.5.2.5.STM where scanning was working with out any issues.

and then we tried mbed os 5.10.4 with latest firmware C3.5.2.5.STM results scan fail issue.

Thanks & Regards
Praveen.Thota

jeromecoutant, 2019-04-12:
Could you check https://github.com/jeromecoutant/wifi-ism43362/tree/PR_SCAN
?

jeromecoutant, 2019-04-24:
Hi @thota4u 
Did you have time to check the change ?

jeromecoutant, 2019-06-19:
Repoened as #57 is not merged

closed 2019-07-03 by jeromecoutant

dansitu, 2020-01-20:
Hi @jeromecoutant - I'm using the latest code but still seeing the issue while using the STM B-L475E-IOT01A1 dev board. I was able to resolve the issue by placing my board in the refrigerator, which sufficiently reduced the number of networks within range.

jeromecoutant, 2020-09-09:
Hi @dansitu 
I have updated a little the scan procedure again in
https://github.com/ARMmbed/wifi-ism43362/pull/83 ?
Maybe you could checked it ?
Thx

@michalpasztamobica

dje666, 2025-08-01:
Dear Forum,

I can't get the WiFi to connect on my ST L475E board either, and it can only detect 8 discrete SSID's. :(