ARMmbed/wifi-ism43362 #63: Regression, second TCP connection closes the first TCP connection

marcuschangarm, 2019-09-23:
I'm running the latest Pelion Device Management client together with a custom application that uses TCP. After Pelion client is connected I start my application. Both Pelion and my application use DNS.

Before commit https://github.com/ARMmbed/wifi-ism43362/commit/7d1e2f9a516614e4090d7fc11a95c00e3d4366b9 everything worked fine, both applications could connect and send data concurrently. After the commit, Pelion client loses connection at exactly the same time my application starts. My application runs fine but Pelion client fails to reconnect.

Platform: DISCO_F413ZH with latest wifi firmware.
Compiler: GCC

LMESTM, 2019-09-24:
First thing I need to ask:
C3.5.2.5.STM firmware version is the latest one - do you confirm this is the version you use ?
In case update is needed, there is a small (Windows) utility and readme here:
https://www.st.com/resource/en/utilities/inventek_fw_updater.zip

LMESTM, 2019-09-24:
@marcuschangarm would you be able to activate traces in ISM drivers files [here](https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ISM43362.cpp#L22) and [here](https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362Interface.cpp#L33) and reproduce your issue ?

marcuschangarm, 2019-09-24:
@LMESTM I've sent you an email with the trace.

jeromecoutant, 2019-10-02:
@michalpasztamobica @SeppoTakalo as regression could come from #61

jeromecoutant, 2020-03-10:
Hi

> I'm running the latest Pelion Device Management client together with a custom application that uses TCP.

Does Pelion client use TCP as well or UDP ?

@JeanMarcR

JeanMarcR, 2020-03-10:
Pelion client can use either TCP or UDP. The choice is done during registration.
You can specify TCP or UDP in the configuration file : mbed_cloud_client_user_config.h
(for instance TCP connection : #define MBED_CLOUD_CLIENT_TRANSPORT_MODE_TCP).

janjongboom, 2020-03-10:
I'm seeing this too but with two connections. I assumed it by design of the WiFi radio?

JeanMarcR, 2020-03-13:
Hi all I would like to run a Pelion Device Management client. But I obtain the following error message: client_error(16) -> Failed to read credentials from storage.

JeanMarcR, 2020-03-23:
Hi all,
I'm running the latest Pelion Device Management client together with a custom application that uses TCP. I'm obtaining a system error message because the ism43362 thread stack size is too small (1024 bytes). I'm increasing the stack size up to 2048 bytes -> no crash.
By default the ism43362 thread stack size = 4096 bytes.
It is possible to change this value (parameter "ism43362.read-thread-stack-size" ) in mbeb_app.json. 

Platform: DISCO_L475E with latest wifi firmware.
Compiler: GCC

Best regards

JeanMarcR, 2020-04-01:
Hi all,
I'm joining an example of two TCP/IP connections :
- the first one is a connection to the PELION cloud,
- the second one is a connection to the echo server.
Two threads (PELION thread - echo thread) are managing the connections.
You can look at the code at the following address:
**https://github.com/JeanMarcR/mbed-os-example-wifi/tree/TWO_THREADS**

Platform: DISCO_L475E with latest wifi firmware.
Compiler: GCC

This example is specifying a problem with the WiFi modem.
Indeed, a valid AT request (R0) is receiving an error message from the WiFi modem.
Best regards.

jeromecoutant, 2021-06-22:
Closing as a workaround has been provided

closed 2021-06-22 by jeromecoutant