ARMmbed/wifi-ism43362 #64: Library dont work with latest Inventek firmware C3.5.2.5.STM

cornebester, 2019-09-27:
When used together with https://github.com/ARMmbed/mbed-os-example-wifi

LMESTM, 2019-09-27:
please provide more information, including mbed-os-example-wifi version and a description of what doesn't work, including a log typically

cornebester, 2019-09-27:
Tried the default version from [here](https://os.mbed.com/teams/ST/code/mbed-os-example-wifi/) using Mbed online as well as updating libs via online. Also tried mbed studio with default version which gave same result and updating to latest libs mbed OS cause build to fail.
Board: DISCO_L475VG_IOT01A 

terminal output

```
WiFi example

Scan:
0 networks available.
No WIFI APNs found - can't continue further.
```
Sample https://os.mbed.com/teams/ST/code/DISCO_L475VG_IOT01-wifi_http_server/ using DISCO_L475VG_IOT01A_wifi.lib works on the latest firmware

LMESTM, 2019-11-28:
Hi @cornebester 

Sorry for not coming back earlier to you. 
I just tested now. With the latest firmware ISM43362-M3G-L44-SPI-C3.5.2.5.STM
I cloned the latest version of mbed-os-example-wifi
I used below mbed_app.json.
Not sure if the idle thread size is needed, but I had this from previous old tests - maybe this is the problem you were having ... 

```
{
    "config": {
        "wifi-shield": {
            "help": "Options are internal, WIFI_ESP8266, WIFI_ISM43362, WIFI_IDW0XX1",
            "value": "internal"
        },
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"LolAp\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"xxxxxx\""
        },
        "idle-thread-stack-size": {
            "help": "Idle stack size",
            "value": "600"
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true
        },
        "DISCO_L475VG_IOT01A": {
            "wifi-shield": "WIFI_ISM43362"
        }
    }
}
```
and this works ok !
```
WiFi example
Mbed OS version 5.14.0

Scan:
Network: NETGEAR_MBED secured: WPA2 BSSID: A0:63:91:31:47:61 RSSI: -66 Ch: 1
Network: MBED_WIFI_TEST secured: None BSSID: C8:D3:A3:67:7d:37 RSSI: -62 Ch: 2
Network: STWLAN2 secured: Unknown BSSID: B0:8B:CF:87:ab:e0 RSSI: -86 Ch: 1
Network: STGUEST secured: None BSSID: B0:8B:CF:87:ab:e1 RSSI: -87 Ch: 1
Network: STSMARTMOBILE secured: Unknown BSSID: B0:8B:CF:87:ab:e2 RSSI: -85 Ch: 1
Network:  secured: WPA2 BSSID: 28:24:FF:8f:7b:c6 RSSI: -90 Ch: 1
Network: LolAp secured: WPA2 BSSID: B2:6F:E0:2f:cf:96 RSSI: -12 Ch: 6
Network: STWLAN2 secured: Unknown BSSID: B0:8B:CF:87:85:40 RSSI: -60 Ch: 9
Network: STGUEST secured: None BSSID: B0:8B:CF:87:85:41 RSSI: -61 Ch: 9
Network: STSMARTMOBILE secured: Unknown BSSID: B0:8B:CF:87:85:42 RSSI: -61 Ch: 9
Network: WIFI_GUEST secured: WPA2 BSSID: 18:64:72:71:9f:71 RSSI: -91 Ch: 11
Network: freebox_0F62E8 secured: Unknown BSSID: E4:9E:12:f:62:e9 RSSI: -96 Ch: 11
Network: FreeWifi_secure secured: Unknown BSSID: 14:C:76:a4:e7:e1 RSSI: -89 Ch: 11
Network: FreeWifi_secure secured: Unknown BSSID: F4:CA:E5:b9:5a:22 RSSI: -96 Ch: 11
Network: Balla secured: Unknown BSSID: 14:C:76:a4:e7:df RSSI: -89 Ch: 11
15 networks available.

Connecting to LolAp...
Success

MAC: C4:7F:51:8C:9D:F6
IP: 192.168.43.169
Netmask: 255.255.255.0
Gateway: 192.168.43.1
RSSI: -8

Done
```

LMESTM, 2019-11-28:
Closing for now - please re-open if you're reproducing the issue

cornebester, 2019-12-14:
Thanks @LMESTM. It works! Appreciated

closed 2019-12-14 by cornebester