ARMmbed/wifi-ism43362 #66: socket_sendto hangs when invalid SocketAddress is provided (only with IAR compiler)

michalpasztamobica, 2019-11-27:
I wrote a simple app to reproduce the issue:
```
#include "mbed.h"

int main (void) {
    printf("Hello\n\r");

    NetworkInterface *net = NetworkInterface::get_default_instance();
    nsapi_error_t result = net->connect();
    
    UDPSocket socket;
    socket.open(net);

    SocketAddress addr;
    result = socket.sendto(addr, "hello", 5);
    printf("result: %d\n\r", result);

    net->disconnect();
}
```

The `addr` is incorrect. For ARM compiler we get a reasonable error code, but for IAR the application hangs and never returns.

See full logs (with debug prints) below.
I see that the modem asks for FW update, but I haven't found any newer firmware than the one we have. I searched [here](https://www.inventeksys.com/ism43362-m3g-l44-at-command-set-firmware-releases/).

IAR:
```
Hello
ISM43362: Reset Module
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.3.BETA9,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3523
ISM43362Interface: please update FW
ISM43362: connection_status 3
ISM43362Interface: update_conn_state_cb 2 -> 3
ISM43362: connection_status 1
ISM43362Interface: update_conn_state_cb 3 -> 1
ISM43362Interface: connect_status 0
ISM43362: receivedIPAddress: 192.168.66.53
ISM43362Interface: socket_open id=0 proto=1
ISM43362Interface: socket_attach id 0
```

ARM:
```
Hello
ISM43362: Reset Module
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.3.BETA9,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3523
ISM43362Interface: please update FW
ISM43362: connection_status 3
ISM43362Interface: update_conn_state_cb 2 -> 3
ISM43362: connection_status 1
ISM43362Interface: update_conn_state_cb 3 -> 1
ISM43362Interface: connect_status 0
ISM43362: receivedIPAddress: 192.168.66.53
ISM43362Interface: socket_open id=0 proto=1
ISM43362Interface: socket_attach id 0
ISM43362:
ISM43362: Buffer content ==========
ISM43362: open: P3 issue
result: -3012
ISM43362: connection_status 2
ISM43362Interface: update_conn_state_cb 1 -> 2
ISM43362Interface: socket_attach id 0
ISM43362Interface: socket_close, id=0
ISM43362: CLOSE socket id=0
```

LMESTM, 2019-11-28:
Hi @michalpasztamobica 

The utility to upgrade firmware on B-L475-IOT01A board can be found here: https://www.st.com/resource/en/utilities/inventek_fw_updater.zip

It also contains instructions and address where to get firmware : https://www.inventeksys.com/iwin/firmware/

The recommended firmware version is ISM43362-M3G-L44-SPI-C3.5.2.5.STM

I will update the README with this information.

michalpasztamobica, 2019-11-28:
Hi @LMESTM ,
Thank you for this information, I will check if firmware update helps. 
In case you have an updated chip at your disposal and if you have time to compile and verify this behavior with IAR, I'd be grateful for feedback.

LMESTM, 2019-11-28:
@michalpasztamobica can you share the mbed_app.json you're using ?

michalpasztamobica, 2019-11-28:
This was enough to run the app I pasted above:
```
{
    "config": {
    },
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "WIFI",
            "nsapi.default-wifi-ssid": "\"SSID\"",
            "nsapi.default-wifi-password": "\"PASSWORD\"",
            "nsapi.default-wifi-security": "WPA2"

        },
        "DISCO_L475VG_IOT01A": {
            "nsapi.default-wifi-security": "WPA_WPA2",
            "ism43362.provide-default": true
        }
    }
}
```
Just set SSID and PASSWORD to your network credentials.
Also, please remember you need to `mbed add wifi-ism43362` in the mbed-os directory.

LMESTM, 2019-11-28:
@michalpasztamobica 
I could reproduce the issue - #67 should solve it.

closed 2019-11-28 by jeromecoutant