ARMmbed/wifi-ism43362 #72: Fails to connect in crowded Wi-Fi environments

JanneKiiskila, 2020-01-22:
Our office has easily 50 Wi-Fi access points. In this environment the Wi-Fi module (with updated firmware) fails to connect.

Not expecting a fix, but possibly should be added as a known issue.

MarceloSalazar, 2020-01-22:
@ARMmbed/team-st-mcd can you please have a look and comment?

ABOSTM, 2020-01-22:
Hi @JanneKiiskila,
It looks like you have the right version (3525) to be able to scan AP one by one, 
but can you verify that you compile with the corresponding wifi-ism43362 patch below ?
https://github.com/ARMmbed/wifi-ism43362/pull/69

janjongboom, 2020-01-22:
Me and @dansitu see the same.

ABOSTM, 2020-01-23:
@janjongboom and @dansitu 
Do you have wifi Firmware version 3525 and do you compile wifi-ism43362 driver with patch #69 ?

JanneKiiskila, 2020-01-23:
In our use case we do not even use scan at all, we know the APN we want to connect to, so we just connect directly to it. So, I'm not sure if the patch #69  would help, since we're not scanning (as far as I know, I don't know then if the driver does some scanning anyway underneath the connect -call?).

LMESTM, 2020-01-23:
Ok that might be an RF issue then ... as you may have seen the board doesn't have a big antenna and I don't think it is very good in long ranges or maybe loudy environments - could you check what happens if the the board is close to the Access Point you're trying to connect to ?

janjongboom, 2020-01-23:
@LMESTM We also see the exact same thing as @JanneKiiskila. Just no replies from the chip if there's lots of networks. But I haven't tried the supplied patch nor updating my WiFi radio.

JanneKiiskila, 2020-02-12:
RF should not be an issue, I've tried to use my mobile phone as the APN, so it's been literally centimetres away from the board itself and it still can't connect.

jeromecoutant, 2021-06-22:
Is this issue still present with all latest patch and up to date wifi FW ?
Thx

closed 2022-07-22 by jeromecoutant