ARMmbed/wifi-ism43362 #74: -3009 (DNS Failed), any idea how to debug?

janjongboom, 2020-05-02:
Hi @jeromecoutant, one of our users is using this WiFi radio on the ST B-L475E-IOT01A but sees all requests fail with a `-3009` error. Looking at the NSAPI error table in Mbed OS that seems like a DNS error, so I'm suspecting a firewall issue somewhere. Do we have any good idea on how we could debug this properly for this chip?

Some info:

https://forum.edgeimpulse.com/t/uploading-finishes-but-no-data-collected/303/7

jeromecoutant, 2020-05-04:
@JeanMarcR Could you have a look ?
Thx

JeanMarcR, 2020-05-04:
Yes, I am looking at.

JeanMarcR, 2020-05-04:
Hi all,

You could try the following steps :
- Removing the gethostbyname() or getb hostbyname_async() from your code,
- Using the IP address of the server (you would like to connect),
- Checking your application. 
If the problem is due to DNS server, your application must run correctly.
Maybe there is a firewall issue.
So, to enable communication with the DNS server, firewall must authorize  UDP and TCP connections on port 53.

Best regards

jeromecoutant, 2020-06-08:
Hi @janjongboom 
Could you try #78 ?
Thx

jeromecoutant, 2020-08-31:
Any update ?
Thx

janjongboom, 2020-09-25:
@jeromecoutant Sorry, GH notifications overload. I can't test this for the user on an unlanded PR. I'll update the WiFi driver for the firmware once it lands (hard to test I know!).

closed 2021-01-14 by jeromecoutant