ARMmbed/wifi-ism43362 #75: DISCO_L475VG_IOT01A doesn't get DNS response from default mbed-os DNS servers in my home network

kivaisan, 2020-05-08:
DISCO_L475VG_IOT01 does not get DNS response from default mbed-os DNS servers (8.8.8.8 and 104.24.122.146 at least) in my home network. 

According to driver trace, in failure case driver never gets an response:
[DNS_nok.log](https://github.com/ARMmbed/wifi-ism43362/files/4598530/DNS_nok.log)

But if I use my operators own DNS server, system gets the response:
[DNS_ok.log](https://github.com/ARMmbed/wifi-ism43362/files/4598533/DNS_ok.log)

I don't see this problem with any other HW (tested multiple devices, e.g. K64F with ethernet or with ESP8266 wifi).

janjongboom, 2020-05-12:
Related #74.

JeanMarcR, 2020-05-14:
Hi all
Indeed, I am reproducing the problem. Moreover I am not able to connect to any DNS server.
Something could help me.
Please, could you generate the traces again (DNS_ok.log) but after setting AT traces.
You can set AT traces at the beginning of the ATParser.cpp.
#define dbg_on           1
#define AT_DATA_PRINT    1
#define AT_COMMAND_PRINT 1
#define AT_HEXA_DATA     1
Thanks
Best regards.

kivaisan, 2020-05-14:
@JeanMarcR Here you go: 
[DNS_ok_at_traces.log.zip](https://github.com/ARMmbed/wifi-ism43362/files/4628885/DNS_ok_at_traces.log.zip)

JeanMarcR, 2020-05-14:
Thanks.
Could you tell me the exact string you give to function gethostbyname() ?

Best regards

kivaisan, 2020-05-14:
This is from [mbed-os-example-sockets](https://github.com/ARMmbed/mbed-os-example-sockets), so string is `"ifconfig.io"`. See https://github.com/ARMmbed/mbed-os-example-sockets/blob/2c14bec4063bdee6b6f4bef6760c8a5c4b5fb7bc/main.cpp#L46

JeanMarcR, 2020-05-18:
Hi,

Please could you set AT traces at the beginning of the ATParser.cpp.
#define dbg_on 1
#define AT_DATA_PRINT 1
#define AT_COMMAND_PRINT 1
#define AT_HEXA_DATA 1

and AFTER
generate again the file DNS_nok.log (in failure case driver).

Best regards

kivaisan, 2020-05-18:
NOK case with AT traces (I stopped test after some time, so example still keeps trying):
[DNS_nok_at_traces.log.zip](https://github.com/ARMmbed/wifi-ism43362/files/4643938/DNS_nok_at_traces.log.zip)

JeanMarcR, 2020-05-26:
@kivaisan 
Please, could you check PR #78

kivaisan, 2020-05-26:
@JeanMarcR #78 works for syncronous DNS queries, but PR is missing async API, which is used for example by PDMC. So please implement `gethostbyname_async` and `gethostbyname_async_cancel` as well.

JeanMarcR, 2020-05-27:
@kivaisan
Functions gethostbyname_async() and gethostbyname_async_cancel() are already present. : NetworkInterface::gethostbyname_async() and NetworkInterface::gethostbyname_async_cancel().
I am also running the test mbed test -t ARM -v -n tests-netsocket-dns without any problem. (test case results: 16 OK)
**So, did you find a problem with these functions ?
or are you surprised not to find them in ISM43362Interface.h**

Best regards

kivaisan, 2020-05-27:
@JeanMarcR Yes they are, but since you now workaround the issue by overriding `NetworkInterface::gethostbyname` by offloading mbed-os' default DNS API functionality into modem's own DNS API, the same fix does not apply for asynchronous API. So if application (like PDMC) calls `gethostbyname_async()`, it still doesn't work.

JeanMarcR, 2020-05-28:
@kivaisan
I am trying to add gethostbyname_async() into modem's API.
It seems more complicated than for gethostbyname(). 
A solution could be to add dynamically (NetworkStack::add_dns_server()) a new DNS server to the list of mbed-os' default DNS servers. This new DNS server is the main DNS server given by the WiFi Network Settings.
What do you think about that ?
Thanks.

kivaisan, 2020-06-01:
@JeanMarcR Please check for example our cellular example how offloaded DNS has been implemented: https://github.com/ARMmbed/mbed-os/blob/c7759fe505d922fe00ee3128dc60110f66647815/features/cellular/framework/targets/QUECTEL/BG96/QUECTEL_BG96_CellularStack.cpp#L482 . I'm not familiar with ISM43362 AT interface, so I can't comment how easy it is to adapt similar solution with it.

I'm not that confident that adding DNS server address would resolve this issue properly. Is it just a pure luck that other DNS server works and another not? In order to properly fix this issue, it would require that it is understood why DNS query is dropped in the first place.

JeanMarcR, 2020-06-01:
@kivaisan

Function QUECTEL_BG96_CellularStack::gethostbyname_async() is written by QUECTEL company. QUECTEL company is the designer of the BG96 : LTE/GPRS Hardware module.

I am not sure that the writing of ISM43362Interface::gethostbyname_async()
concerns STMicroelectronics. 
STMicroelectronics is not the designer of the WiFi HW module : ISM43362.

Regards

kivaisan, 2020-06-01:
@JeanMarcR I was able get tcpdump from my router and I think I found the reason for the issue. ISM seems to add the same port (53) as source port to DNS UDP packet and I think my operator will block that response from internet.

Please check if you can modify that source port to be some >1024. Although it is most probably set by IP stack in ISM43362 module, so fix should be requested from Inventek.

JeanMarcR, 2020-06-02:
@kivaisan
The port (53) of DNS server is specified (hard coded) in nsapi_dns_get_server_addr() function. This function belongs to mbed-os module not ISM43362.
You can easily modify the value but you need to re-compile mbed-os.
Regards.

kivaisan, 2020-06-02:
@JeanMarcR Yes, but that is destination port. I'm talking now about source port.

ISM driver seems to set the source port here: 
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ISM43362.cpp#L677-L696 
and according to my above NOK log, port is something else but 53: `AT> P2=64779` , but still according to tcpdump, source port is 53.

JeanMarcR, 2020-06-02:
@kivaisan
I think the local port is randomly defined in ISM43362::open().
In my traces I can find : P2=59437 or P2=59436 ... with OK returned by the WiFi module.
Be aware also that it is the UDP protocol which is used and not TCP protocol.
The only explanation could be that the P2=59437 is not well interpreted by the WiFi module.
Where is the TCPDUMP generated ?

kivaisan, 2020-06-02:
Maybe there is a bug then in ISM wifi firmware.
I took tcpdump from my home router to see what data goes through it.

JeanMarcR, 2020-06-02:
@kivaisan
I am going to write a mail to our WiFi module supplier (INVENTEK SYSTEMS) to explain the problem : incoherence between the local port number (64779) given at AT command P2  and the local port number (53 ) given by your tcpdump traces.

JeanMarcR, 2020-06-05:
@kivaisan

Please,
Could you send me the full traces ?
Best regards

kivaisan, 2020-06-05:
What do you mean by "full traces"?

JeanMarcR, 2020-06-05:
@kivaisan
You spoke about NOK log and TCPDUMP from your HOME network.
Best regards

kivaisan, 2020-06-08:
@JeanMarcR Here is one run with atparser log and screenshot of the tcpdump (I'm not sharing the while tcpdump capture as it contains a lot of other network data):
[dns_nok_2.log](https://github.com/ARMmbed/wifi-ism43362/files/4744045/dns_nok_2.log)
[image]

JeanMarcR, 2020-06-15:
@kivaisan
The Inventek approach concerning UDP socket is quite confusing.
 P2 is local Port and P4 is remote port according from Inventek AT command . 
**This notion becomes false for UDP approach** .
To send UDP data from inventek to a remote you have first to start what inventek is calling an UDP client. While setting this client, you have to provide the local socket you want to use and you should **use P4 command for this purpose and not P2** .
So it’s a local port but must be provided though P4 .

So here the command for UDP client : 

#select socket
> P0=0
OK
#select UDP
> P1=1
OK
#select Remote Addr
>P3=192.168.1.65
OK
**#select local port
> P4=1234**

OK
#start udp client
> P6=1
[UDP  RC] Socket opened on port 1234
OK

Then you can start to send  UDP data
#select socket
> P0=0
OK
#select remote destination port
> P4=54
OK
#send data
> S3=4\r1234
4
OK
#at that stage the remote connection received data on port 54 from inventek port 1234

Best regards.

JeanMarcR, 2020-06-15:
@kivaisan
I am writing a patch.

kivaisan, 2020-06-15:
@JeanMarcR Good that this got clarified. Above doesn't look confusing at all. It is normal that command set is different based on protocol.

JeanMarcR, 2020-06-19:
@kivaisan
This patch could solve the problem.

[patch.txt](https://github.com/ARMmbed/wifi-ism43362/files/4802905/patch.txt)

kivaisan, 2020-06-22:
@JeanMarcR Tnx. Now it DNS works nicely.

But I would change `type` parameter to be `nsapi_protocol_t` instead of char and doing to char conversion inside `ISM43362::open` for `P1` parameter. This would simplify the code and get rid of unnecessary `memcmp` operations.

jeromecoutant, 2020-06-26:
@kivaisan 
Let's do it in 2 steps ?
- DNS patch in this PR as proposed by @JeanMarcR 
- API update ?

kivaisan, 2020-07-02:
What ever way suites best for you. But could you please submit the fixes?

JeanMarcR, 2020-07-02:
@kivaisan
Yes I am going to submit the fixe.

jeromecoutant, 2020-08-31:
I suppose it is #78

jeromecoutant, 2021-01-14:
Is #78 OK to close this issue ?

kivaisan, 2021-01-14:
Yes. I close the issue.

closed 2021-01-14 by kivaisan