ARMmbed/wifi-ism43362 #76: Driver returns 0 for UDP socket recvfrom if called before sendto

kivaisan, 2020-05-18:
If user has and UDP socket and first calls `recvfrom` before calling `sendto`, driver always returns data size of 0.
This is a bit confusing. If driver has no data, maybe it should return `NSAPI_ERROR_WOULD_BLOCK`?

jeromecoutant, 2020-05-18:
@JeanMarcR

JeanMarcR, 2020-10-19:
@kivaisan
Hi

Do you use the board B-L475E-IOT01A ?
 
Best regards

kivaisan, 2020-10-19:
Yes. If I remember correcttly, the problem is in this part:
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362Interface.cpp#L479-L482

Internal connect flag is not set until user calls `sendto`.

JeanMarcR, 2020-11-02:
@kivaisan
Hi
1 - ) socket.open()
2 - ) socket.connect()
3 - ) socket.recv() -> return 0

Please, could you tell me if it is the sequence of function calls you have done ?
Best regards

kivaisan, 2020-11-02:
I don't remember anymore the use case where I found this, but I think your sequence is right.

JeanMarcR, 2020-11-03:
@kivaisan

Please, could you check the following patch:
in file wifi-ism43362\ISM43362Interface.cpp
in function ISM43362Interface::socket_recv()

replace lines
   if (!socket->connected){
        _mutex.unlock();
	return 0;
    }

by lines

   if (!socket->connected){
        _mutex.unlock();
	if (socket->proto == NSAPI_UDP) {
	   return NSAPI_ERROR_WOULD_BLOCK; 
	}			
	return 0;
    }

kivaisan, 2020-11-04:
@JeanMarcR Sorry, I'm currently working on another project, so I cannot test this right now.

PR #84 (2020-11-13): MODIFYING RECEIVE FUNCTION IN CASE OF NO CONNECTED UDP SOCKET
- Add a new return in socket_recv() :
"return NSAPI_ERROR_WOULD_BLOCK" to take into account a no
connected UDP socket.

fix #76
closed 2020-11-13 by jeromecoutant