ARMmbed/wifi-ism43362 #80: Opening client connection to remote server shows success even if  no server is started on the specified remote IP and port

iampoonam, 2020-06-04:
The AT command P6=1 always gives success even if there is no server with the specified IP address and port. Is there a way to check if the target as client actually got connected to the specified server ?

LMESTM, 2020-06-04:
@iampoonam - I think this question should rather be asked to Inventek support .

han0110, 2020-12-19:
To check whether socket is connected to remote one, my workaround is to inherit class `InternetSocket` and use `ISM43362_socket` to inspect the protected member `nsapi_socket_t _socket` to check whether is connected to remote socket. Like this:

```c++
#include "TCPSocket.h"

// Copy from ISM43362Interface.cpp. Reference: https://github.com/ARMmbed/wifi-ism43362/blob/3813a4bb8623cc9b0525978748581f60d47142fa/ISM43362Interface.cpp#L301-L308
struct ISM43362_socket {
  int id;
  nsapi_protocol_t proto;
  volatile bool connected;
  // SocketAddress addr;
  // char read_data[1400];
  // volatile uint32_t read_data_size;
};

class TCPClient : protected TCPSocket {
  bool connected() { return ((struct ISM43362_socket*)_socket)->connected; }
}
```

jeromecoutant, 2020-12-21:
Hi
If you have some good workaround that can be shared with all,
don't hesitate to propose a pull request :-)
Thx

closed 2021-06-22 by jeromecoutant