ARMmbed/wifi-ism43362 #87: Floating point usage

LDong-Arm, 2020-11-25:
Mbed OS aims to be free from floating point usages. This driver brings in some floating point symbols:

```
$ python mbed-os/tools/test/examples/elf_float_checker.py ./BUILD/DISCO_L475VG_IOT01A/ARM/mbed-os-example-for-aws.elf 
Checking ./BUILD/DISCO_L475VG_IOT01A/ARM/mbed-os-example-for-aws.elf for floating point symbols
Failed - Found float symbols:
__aeabi_d2f
__aeabi_cdcmpeq
__aeabi_cdcmple
__aeabi_dmul
__aeabi_cdrcmple
Found float symbols in ./BUILD/DISCO_L475VG_IOT01A/ARM/mbed-os-example-for-aws.elf
```
Without this library, there's no float symbol.

JeanMarcR, 2020-11-26:
I did not find any "float" or "double" in the wifi module.
Please, could you specify the function concerned by floating point symbols ?

jeromecoutant, 2020-12-04:
Hi
Maybe we can close this issue ?
Thx

LDong-Arm, 2020-12-07:
Thanks, I suspect the symbols might be from either Mbed OS WiFi stack, or the C library.

closed 2020-12-07 by LDong-Arm