ARMmbed/wifi-ism43362 #88: wifi-ism43362 no response on B_L4S5I_IOT01A

andrew1205lin, 2021-02-04:
Greetings,

I tried to connect Wi-Fi on my B_L4S5I_IOT01A, 
with both mbed-os-example-wifi(https://github.com/ARMmbed/mbed-os-example-wifi), 
and mbed-os-example-sockets(https://github.com/ARMmbed/mbed-os-example-sockets),
but they didn't work.

## Here are some details:
### Running mbed-os-example-sockets directly

[image]

failed.

### Modifying mbed_app.json in mbed-os-example-sockets because it wasn't including the "B_L4S5I_IOT01A" block at the beginning.

[image]

### Rerunning with debug again

[image]

This time, it seems that B_L4S5I_IOT01A sent a AT command "I?" to wifi-ism43362, and got no response.

## Firmware
I've upgrade my firmware ISM43362-M3G-L44-SPI-C3.5.2.5.STM with inventek_fw_updater.zip,
(https://www.st.com/resource/en/utilities/inventek_fw_updater.zip)
but I cannot tell if it was upgraded with no mistake, 
since I don't know how to check the firmware version of B_L4S5I_IOT01A.

That's all, I have no idea what has gone wrong. 
Please guide me what I can do.

JeanMarcR, 2021-02-04:
Please, could you send me your mbed_app.json ?
Thanks

JeanMarcR, 2021-02-04:
@andrew1205lin
Please, could you also send me the text present in the Tera Term ?

andrew1205lin, 2021-02-05:
@JeanMarcR 

My mbed_app.json. It is the original mbed_app.json in the mbed-os-example-sockets, with "B_L4S5I_IOT01A" added.
```
{
    "config": {
        "hostname": {
            "help": "The demo will try to connect to this web address on port 80 (or port 443 when using tls).",
            "value": "\"ifconfig.io\""
        },
        "use-tls-socket": {
            "value": false
        }
    },
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "WIFI",
            "nsapi.default-wifi-security": "WPA_WPA2",
            "nsapi.default-wifi-ssid": "\"0206-3\"",
            "nsapi.default-wifi-password": "\"12345678\"",
            "platform.stdio-baud-rate": 9600,
            "mbed-trace.enable": false,
            "mbed-trace.max-level": "TRACE_LEVEL_DEBUG",
            "rtos.main-thread-stack-size": 8192
        },
        "DISCO_F413ZH": {
            "target.components_add": ["ism43362"],
            "target.macros_add" : ["MBEDTLS_SHA1_C"]
        },
        "DISCO_L475VG_IOT01A": {
            "target.components_add": ["ism43362"],
            "ism43362.provide-default": true,
            "target.network-default-interface-type": "WIFI",
            "target.macros_add" : ["MBEDTLS_SHA1_C"]
        },
        "B_L4S5I_IOT01A": {
            "target.components_add": ["ism43362"],
            "ism43362.provide-default": true,
            "target.network-default-interface-type": "WIFI",
            "target.macros_add" : ["MBEDTLS_SHA1_C"]
        },
        "K64F": {
            "target.macros_add" : ["MBEDTLS_SHA1_C"]
        }
    }
}

```

As for text present in the Tera Term, I didn't use Tera Term. Which text do you want? In which step?

JeanMarcR, 2021-02-09:
@andrew1205lin 
I am reproducing the problem on my board. 
To deepen the problem I must use the debugger MbedStudio which does not suggest the target STM32L4S5I. So, I am going to add STM32L4S5I to MbedStudio and after debugging your problem.

andrew1205lin, 2021-02-09:
@JeanMarcR 
I have used another board, DISCO-L475VG-IOT01A, to follow the same procedure and do the same example. 
Then, it produced the same problem, ism43362 return two lines of "/r/n".
After this attempt, I think I didn't update my firmware correctly from the beginning.

JeanMarcR, 2021-02-11:
Hi

I made a test with the following WIFI module:
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.                                                                             2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3525
with board **DISCO-L475VG-IOT01** **mbed-os 6.7.0**

As you can see the result of get_firmware_version() is not FAIL as in your console.
No crash appears.
I used the joined mbed_app.json file.
[mbed_app.txt](https://github.com/ARMmbed/wifi-ism43362/files/5965606/mbed_app.txt)

JeanMarcR, 2021-02-15:
@andrew1205lin

Please, could you give me the result of command : **mbedls** for boards L475 & board L4S5 ?

Thanks

JeanMarcR, 2021-02-15:
Hi,
I run with no error the mbed-os-example-sockets application on the STM32-L4S5I board.
I needed to rebase my git repository of WiFi Library thanks to the master of https://github.com/ARMmbed/wifi-ism43362.git
You could check the presence of board B_L4S5I_IOT01A in mbed-os-example-sockets\wifi-ism43362\mbed_lib.json.

Best regards

andrew1205lin, 2021-02-15:
Hi,

Here's the result of command :  **mbedls**.
[image]

Thanks

JeanMarcR, 2021-03-05:
@andrew1205lin 
May I close the issue ?

andrew1205lin, 2021-03-05:
@JeanMarcR 
OK, you can close this issue.
Thanks for your support.
I think there may be some silly mistakes I have made.

Final question, could you tell me if I need to do something extra, such as move the jumper, when updating firmware, please?

JeanMarcR, 2021-03-09:
@andrew1205lin 
No extra things to do to update the WiFi firmware.
You can find the procedure : https://github.com/ARMmbed/wifi-ism43362
§Firmware version & firmware update

closed 2021-03-09 by JeanMarcR